juhakaup/keyboards #4: Would it be possible to get the fusion files?

Nobodi99, 2023-01-07:
Hi,
I recently printed one half of your design to test and I really like the aesthetics but I would love to move the upper pinky key above the others and raise the innermost thumbkey because the middle key gets in the way for me.
Would it be possible to get access to the design files to modify it?

juhakaup, 2023-01-15:
Glad to hear you like the design.
Here is the link to the fusion files:
https://drive.google.com/file/d/1OVxvqivmw3vvyvD0ghFZgAA_TH4tXXlT/view?usp=sharing
Modifying the design may be quite challenging.

I'm working on a new version that is parametric and should be easier to modify.
I've uploaded a layout testing file to this repocitory, you may want to check it out as well.

edte, 2023-10-15:
Have you changed it? I also want to change the keys on my little finger, including the q key and p key.

juhakaup, 2023-10-15:
I've updated the repository, fusion files are now included. This version is completely rebuild and has parameters for key positions etc. I also included a version where the P and Q keys are with the other pinky keys.
I'll add documentation at some point, when I have time.

closed 2024-03-03 by juhakaup

weilbith, 2024-03-04:
Thank you very much @juhakaup 🙏🏾 
I loaded your project into Fusion 360 and looked into the history to learn which construction approach you chose. The key part for me was to create an object and cut out the key holes. My approach was connecting the key holes and then trying to build a case around it, connecting to the object faces. Etc. Was/is a pita and broke everytime the key whole layout changes due to prototypes. Will learn now how to construct objects properly.
This finally gives me motivation and hope again that I can "finish" my own keyboard vision.