juhakaup/keyboards #12: Any chance for a 4x6?

shensmobile, 2024-04-12:
I know it might ruin the design language of the board, but is there any chance you might consider making a version of this board with a numrow? I’m one of those ergo enjoyers that prefers a little bit less minimalism.

hailee0710, 2024-08-04:
Thumbs up for this, I really love to see the 4x6 version, can't live with the numrow

juhakaup, 2024-08-07:
[image]
[image]

Here's a sneak peek.

hailee0710, 2024-08-08:
wow, fabulous!
[image]

marioolofo, 2024-08-14:
Very nice @juhakaup 
Can you share the model to print?

shensmobile, 2024-08-16:
Absolutely fantastic.  I can't wait to print it :D

juhakaup, 2024-08-25:
4x6 is released.

closed 2024-08-25 by juhakaup