juhakaup/keyboards #14: Unresolved components in Fusion360 source file

justanothernguyen, 2024-04-12:
[image]

The missing components prevent further editing

juhakaup, 2024-04-13:
All right. Can you be more specific, what file, have you made any changes? Does the folder contain the switchholder -component?

justanothernguyen, 2024-04-13:
Sorry about this, I imported [Cygnus v1.0](https://github.com/juhakaup/keyboards/blob/main/Cygnus%20v1.0/fusion360/cygnus.f3z) and opened the cygnus_case component.

No 
changes were made. Oddly enough after checking I do find the Switch, switchholder etc. components imported.

juhakaup, 2024-04-14:
Haven't really encountered this issue before. The files seem to work fine for me.
You haven't decompressed the file or something?
You possibly solve the issue by using the replace component function from the right click menu on one of the defective components.