juhakaup/keyboards #15: Battery position?

laporchen, 2024-04-14:
I current making a wireless version of 3x5 v1.

You mentioned about using double side duct tape to keep battery in place.
I wonder if you can provide an example of the interior for wireless version. 
I can't tell where should the position for battery be since I haven't got all the parts yet and the interior is very small.

Thank you in advance.

juhakaup, 2024-04-14:
Here is how I attached the battery.
It's about 30 x 20 x 7mm in size.
[image]

laporchen, 2024-04-14:
Thanks for the reference.
Btw, what kind of material did you use for this case, the color is magnificent.

juhakaup, 2024-04-14:
It's Burnt Titanium PLA by Flashforge.

fredrikjacobson, 2024-05-29:
What's the clearance like for the battery? I wonder if a larger one would fit or if this is already a tight fit. @laporchen what battery did you end up using?

juhakaup, 2024-05-30:
I have some 550mAh 503040 batteries that theoretically fit in the cavity, but would require fabricating some sort of stand to get them to stay in the odd position. The 250mAh 502030 batteries I have seem to last quite a while and fit in the case without too much trouble.

dudicoco, 2024-07-04:
@juhakaup what is the clearance in mm from the the bottom of the case to the switches pcbs?

juhakaup, 2024-07-05:
Well it's not quite that simple. The keywell is concave and I made the keyboard as low as possible, so some of the pcbs have pretty much no clearance at all. There really isn't all that much space inside the case, so you may need to a quite creative to fit a large battery.