juhakaup/keyboards #16: controller clamp mirrored thread is left-handed

georgewild42, 2024-04-15:
When mirroring the controller clamp in slicer, the thread becomes left-handed. Am I missing something here?

[image]

I would fix this myself by making the thread left-handed in Fusion before mirroring, however the supermini wireless bracket is missing from the Fusion file.

Thank you for this design, it's great. I am looking forward to building the wireless 3x6 as my first ergo.

juhakaup, 2024-04-21:
Yes this is an clear oversight on my behalf. In practice, the resolution of the 3d printed m3 thread is not that great, so it doesn't really affect the performance.

georgewild42, 2024-04-21:
Yeah, you are right it doesn't really matter. For anyone else with OCD however, I edited the STL file for the supermini wireless:

[controller_holder_supermini_wireless v1 LEFT HANDED.zip](https://github.com/juhakaup/keyboards/files/15051466/controller_holder_supermini_wireless.v1.LEFT.HANDED.zip)

closed 2024-04-21 by georgewild42