juhakaup/keyboards #19: Case Doesn't Close Properly Due to Switch Socket Position 3x6

minchik, 2024-04-27:
When attempting to close the 3x6 version of the case, the switch socket's proximity to the bottom of the case prevents proper closure. This issue is caused by the socket's position being too close to the lower edge, resulting in an incomplete or forced closure.

Can you increase the height of the bottom part by 2mm?

[image](https://github.com/juhakaup/keyboards/assets/2690703/587bae05-a9e4-4f69-b846-4ad9afa0442f)

juhakaup, 2024-04-28:
Don't really want to do that, but I added a notch to the bottom so the socket should fit now. Let me know if it fixed the issue.

minchik, 2024-05-06:
Thanks! I have printed it again, and the issue is fixed.

closed 2024-05-06 by minchik