juhakaup/keyboards #25: 1mm thick walls on bottom risk of breaking when printing?

LarssonMartin1998, 2024-06-01:
Hey,

First I want to apologize if this is not an issue. I have no experience at all when it comes to 3D printing. I am trying to print this keyboard with the help of jlc3dp's services. I uploaded the 3x5 uniform top and the regular bottom, plus the mirrored versions.

They replied with a warning, I'll quote them from here:

"The wall thickness of those indicated red area is too thin, which has high risks of loss, deformation, crack, or damage during the printing process.

Could you please confirm if the risks are acceptable for you?

If not, could you please kindly increase it to be at least 1.0mm to proceed? 2.0 mm will be better."

Here is the image in question:
[image]

I tried to print MJF PA12-HP Nylon, and here is what they said about the different materials:

"Kindly noted resin has the lowest requirements for the wall thickness, which is at least 0.8m and 1.5mm is better. But the risks still exist for the red areas mentioned in the picture.

Kindly noted, nylon is more flexible for resin is brittle. Please refer to the design guideline in this [link](https://jlc3dp.com/help/article/212-3D-Printing-Design-Guideline)"

I wanted to bring this to your attention, but I also wanted to ask if this is okay to print, maybe they're just being overly cautious? Not sure.

evantravers, 2024-06-01:
Hey! I'm pursuing the same path here: https://github.com/juhakaup/keyboards/discussions/23

turns out you can say "pretty please" and accept the risk and they'll print it anyway. I've got one from PCBway coming soon, I'll report back in that thread. If you want, join the discussion on the other thread.

LarssonMartin1998, 2024-06-01:
> Hey! I'm pursuing the same path here: #23
> 
> turns out you can say "pretty please" and accept the risk and they'll print it anyway. I've got one from PCBway coming soon, I'll report back in that thread. If you want, join the discussion on the other thread.

Hey,

Thanks for the suggestion, I got the same sentiment from jlc3dp, they told me that they could still print it so long as they had my written consent that I am willing to accept the risk. However, as the Nylon printing for all of the parts is roughly $80, I wanted to make sure that it's feasible before committing to it.

I'll definitely join your thread! Just a question, what is the rough ETA for your printed case? Wondering if I should wait and see your result, wait for the owner to reply to this issue, go ahead and just learn some basics of modeling so I can increase the width myself, or just go for it.

evantravers, 2024-06-01:
> what is the rough ETA for your printed case

PCBway's production status says 8 days, then shipping to CONUS from China, so.. ~14ish?

I will say, unless you are doing something like Nylon or Resin printing, I think the shipping cost is pretty prohibitive doing PCBway for this. Next time I'm certainly going to use local 3d printer. I think shipping alone was the cost of printing two cases. (~80 USD)

guategeek, 2024-06-01:
I have not printed the Cygnus yet, but I have two 3D printers and have five years of experience 3D printing.

Looking at the image those thin portions are in the bottom plate so even if they were open it wouldn't really matter (its not a structural part of the case). Having even a single layer of plastic (if the print at 0.2mm layer height) there would help keep dust out of the case.

I'm guessing those areas are cut out for clearance issues so modifying the model could make the switches (well the single key pcbs on the switches hit the bottom and cause the case to not close correctly.

I would say you are probably fine getting it printed, and as far as materials I would say go with MJF PA12-HP Nylon (especially if its made with the SLS process) if you can, its going to make for a much stronger case. Resin prints are much more brittle than FDM and SLS prints so if you go that rout be aware that a drop could easily break the case.

LarssonMartin1998, 2024-06-01:
> I have not printed the Cygnus yet, but I have two 3D printers and have five years of experience 3D printing.
>
> Looking at the image those thin portions are in the bottom plate so even if they were [...]

Thank you for the information, I value it very much, means a lot! I'll go ahead and order it in MJF PA12-HP Nylon. I'll give an update here whenever I recieve it, but I'll go ahead and close it.

closed 2024-06-01 by LarssonMartin1998