juhakaup/keyboards #29: Add hole for LED battery indicator

dudicoco, 2024-07-04:
Hi,

Thank you for this great project.

Would it be possible to add a hole for an LED so we can get battery status indication?

Thanks

juhakaup, 2024-07-04:
Yes its possible, I've thought about it myself.
I haven't looked into this too much, so if you have any ideas on how to implement it I would appreciate it. (what kind of hardware to use, does the firmware need to be updated)

If you just need a hole for a led, that can be done in a slicing software quite easily.

dudicoco, 2024-07-04:
@juhakaup i'm still unsure about the exact details of the implementation but I did find this repo: https://github.com/caksoylar/zmk-rgbled-widget

Also found this PR: https://github.com/zmkfirmware/zmk/pull/1120

Would it be possible to add a hole for the LED until we figure out the implementation details? Asking for this since I need to place an order for the 3d printed case.

Thanks

juhakaup, 2024-07-06:
[image]
This is what I came up with. It uses a 3mm led, that should be easily accessible. 
There isn't a lot of room in the case, so I placed the led at an angle to make the installation easier.
I didn't want a large hole in the case, so I made a 1.8mm hole so that you can use something like transparent filament as makeshift lightpipe.

Let me know if you want to try this out, so that I can make the change to the appropriate model and upload it to the requests folder.

dudicoco, 2024-07-16:
Thanks @juhakaup!

I guess I need to be 100% sure that this will print out ok since i'm printing the case over at JLCPCB.

Do you think there could be issues with the print or with the fit and installation of the LED?