juhakaup/keyboards #35: Any chance to get 4x6 step format?

nicesone, 2024-09-10:
Cygnus is the best keyboard ever! I use it at work and home a lot! 

It's so exciting to see the 4x6, is there any chance to have the 4x6 in STEP format?

Many thanks!

nicesone, 2024-09-11:
Let me know if I should open a discussion instead. Thx

juhakaup, 2024-09-11:
I'll see about uploading the step files later this week, when I have time.
Are you planning to modify the design?

nicesone, 2024-09-11:
Awesome!

Not really, I'm asking my friend to make a 4x6 for me. They told me that due to some 3D printer limitations, it would be great if it's in STEP format. I didn't get the full picture, but that's the reason why I'm here :D

juhakaup, 2024-09-14:
I've uploaded the files in step format.
They are in the [requests folder](https://github.com/juhakaup/keyboards/tree/main/Cygnus%20v1.0/requests%20and%20other%20experiments/step%20files).

nicesone, 2024-09-15:
Amazing!

I can't wait to send it to my friends! 

Many thank!!!

closed 2024-09-15 by nicesone