juhakaup/keyboards #42: How to open/edit in Fusion?

222Phoenix, 2024-10-24:
I'm not able to open any of the .f3z files, any ideas? I tried all three, but get this error:

Error: Fusion 360 could not open cygnus_3x6.f3z because it is not a supported file type.

juhakaup, 2024-10-24:
Can't open it directly. You need to use the upload feature to open it.
https://forums.autodesk.com/t5/fusion-design-validate-document/opening-a-f3z-file/td-p/6391844

222Phoenix, 2024-10-24:
Thanks. I definitely tried that also. I click data panel, then upload, choose the file, and upload, but nothing happens.

juhakaup, 2024-10-24:
You might need to create a project first and then select it as location in the upload menu.

222Phoenix, 2024-11-18:
I finally figured this out. I don't get why this makes a difference but it does: on the file, you cannot just right click the file and choose "save as." You must click the "Download raw file" button.

closed 2024-11-18 by 222Phoenix