juhakaup/keyboards #47: Wireless version Fusion File

Dafeigy, 2024-12-10:
Hi, I really like your design and I successfully print them with bambu A1. I am working on 3x5 wireless version and I noticed that there are two irregular grooves at the bottom plates (`/Cygnus v1.0/stl/cygnus 3x5/cygnus_3x5_bottom_wireless.stl`), and it seems very fragile so I was planing to remove them. I am new in fusion360 so I don't know how to edit `.stl` files, so can you provides a wireless version fusion files? Thanks again for your excellent work!

juhakaup, 2024-12-10:
Hi. The holes are there to make room for the hotswap sockets. If you remove them you may have issues with the sockets pushing against the bottom.

If you still want to remove them, or to make them a bit less thin, probably the easiest way is to do it in the slicer. You should be able to add a geometric shape, like a box, through the slicer menus and combine that with the model to solve the issue.

I've printed several cases and haven't had any issues with the wall thickness.

Dafeigy, 2024-12-11:
Many thanks! In that case I will keep those holes :)