juhakaup/keyboards #50: In which orientation should I print the top?

repercussion, 2025-01-17:
Wondering how you have printed the top. I've tried twice using tree supports, but the print has failed both times at the thumb cluster. Perhaps I'm just going too fast. 

Both attempts have been horizontal with the inside of the shell face down. 

The STLs come balanced on the top and I can't make out from the lines in your print's image how you did it.

juhakaup, 2025-01-17:
I usually print them flat, as in just rotate the STL 90°. I haven't had issues with this orientation, but I've heard that others have.
I've also printed them vertically so that the top halve stands on the square at the side of the thumb cluster, this is useful if you want to sand the case to make it smoother.
Speed isn't a huge issue for me when printing these cases. My old prusa printer isn't fast, but it's reliable, so it's kind of set it and come back the next day for me.

repercussion, 2025-01-17:
Very much appreciate the input. The confirmation helps. 

This seemed a bit much:
[image]

I will attempt printing in this orientation again:

[image]

Thank you! 

I will try again with normal supports and slow the print right down.

closed 2025-01-17 by repercussion