juhakaup/keyboards #57: Most corne/crkbd firmware requires two wires for serial communication.

zfralish, 2025-04-30:
Love the keyboard, have built a wireless already and is working great.

Trying to build a wired version as well and I am running into an issue I don't understand.

I followed the wiring diagram from the readme, but all qmk corne firmware seems to require 4 wires.

In my case I have a pro micro (have also tested with elite c) with TX0 - TX0, RX0 to RX0, GND - GND, VCC - VCC. This is the only way to corne firmware works.

I am not sure if this is something misconfigured on my side or if something has changed with respect to the corne firmware since the docs were written. I am just using the default firmware from qmk. I have also tried the legacy, common, and r2g variants.

This is not really an "Issue" as it works fine this way, this is just not how i understand serial communication to work, and would like to understand why this is the case.

juhakaup, 2025-05-01:
Probably a change with the v4 corne.
The documentation for the classic corne still talks about 3 wire trs cable.
https://github.com/foostan/crkbd/blob/main/docs/corne-classic/buildguide_en.md
QMK does support serial communication with 3 or 4 wires, so it's just a matter of configuration. Part of the reason I went with corne firmware was to avoid doing configuration, so I'm probably not the right person to talk about that.
Qmk configurator has crkdb/rev1 that could be for the older version, if you want to check that out.

axhixh, 2025-05-06:
Use of 3 or 4 wires to communicate between the two halves depends on the protocol used. If you use I2C then you will need 4 wires. If you want to use "serial", not that I2C isn't serial, you can use either 3 or 4 wires based on half-duplex or full-duplex mode. It you use bitbang, again you only need 3 wires.

See https://docs.qmk.fm/drivers/serial